clough42/3d-printing #10: 1.75 universal wont fit without modification

Sniffle, 2015-03-09:
The 1.75 universal comes with a very small bowden tube adapter installed that needs a small recess around the 3 mm hole.  included are pictures to show this.

It is roughly 2mm deep,and about 2-3mm larger in diameter than the filament hole.  

image direct from e3d site  same as used on 1.75 universal
http://e3d-online.com/image/cache/data/kraken/E3D_Kraken09thumb-1000x1000.jpg

pics i've taken for reference

side view of adapter protrusion
http://i34.photobucket.com/albums/d130/tourach23/Mobile%20Uploads/2015-03/20150309_075045.jpg

reference pic for area of needed change
http://i34.photobucket.com/albums/d130/tourach23/Mobile%20Uploads/2015-03/20150309_075056.jpg

Sniffle, 2015-03-10:
I used a 5/16th drill bit to countersink about a 16th of an inch and it fit fine.

clough42, 2015-03-10:
From the engineering drawings, it looks like the protrusion is 7mm in diameter and 1.4mm high.

I'm trying to work out exactly how to create the counterbore and still allow the part to be printed without support.

There's a tube to provide filament support, right?  So if the channel isn't perfectly tight, it'll still be okay?  I'm thinking about making an 8mm counterbore above the support webbing for about 1.4mm, and then taper at 40 degrees or so to the channel diameter.  I want to avoid having a second support web in the hole that has to be cut out.

Thoughts?

clough42, 2015-03-10:
Oh...does the tube fit in the bore (for 3mm filament) as printed, or did you have to drill that out as well?

Sniffle, 2015-03-10:
So long as there is space for it to fit the 1.75 has the PTFE liner... Its
the 3mm folks who might have issue with a taper because that's more room
for coiling to occur on something like ninjaflex

Sniffle, 2015-03-10:
I actually forgot to check the PTFE Liner glad you asked, 

It will go in, though it is TIGHT, but as long as it goes i would say that is good enough.

I was laying in bed with my son so not able to respond fully earlier, If you are going to use a bevel to preven another support(which is understandable, I would suggest keeping it as sharp as you think you can get away with, because of the reasons i mentioned above.

Mainly being that any kind of gap, the larger it is the more prone to having filament buckle in that space and jam up.

clough42, 2015-03-10:
Okay.  I'll model something up later this week.  It would be easy to make two versions: one for 3mm and one for 1.75 with the recess and a slightly looser bore.  I need to get one printed and assembled with Hexagons to see how the Ninjaflex responds to it.

Sniffle, 2015-03-10:
I went ahead and built this onto the printer, 

Go ahead when you make the 1.75 version enlarge the hole a smidge to allow for the ptfe tube i had to drill out the holes to allow it to fit... they became wedged and unable to move before reaching the hobbed bolt.

clough42, 2015-03-12:
How big did you have to drill it?  Even just the outside diameter of the tube would be helpful.

Sniffle, 2015-03-12:
Im not at home I'll measure when I get home

Sniffle, 2015-03-12:
The PTFE tube measured 4.0mm exactly

clough42, 2015-03-16:
I just made another release (0.2.0) that has an E3D-specific version with the proper recess and a slightly larger filament channel.  (Double Flex Block-175EC.STL).

closed 2015-03-16 by clough42