clough42/3d-printing #11: E3D V6 Hot end double flex groove mount too tall

Sniffle, 2015-03-10:
I had to shave it down to about .4mm(2 layers left) to be able to fit the E3D into the groove mount plate.

sorry for the unscientific measurements but my kid found and lost my calipers

Pic for reference
http://i34.photobucket.com/albums/d130/tourach23/Mobile%20Uploads/2015-03/20150309_201640.jpg

clough42, 2015-03-10:
I sized the mount based on the delta between the Hexagon and the e3dV6.  It would be really helpful if you could measure the final dimension of the plate after shaving and measure the inside of the groove on the hot end.

I'm wondering if the model is too thick, or if your printer is printing the part a little thicker than mine for some reason.

Sniffle, 2015-03-10:
I'll try to find my calipers and jump on this... Its literally like .2mm...
It just jammed up to the point where I couldn't seat it or I wouldn't have
said anything

Sniffle, 2015-03-10:
Calipers Found!(i blamed my kid falsely they fell off the top of my enclosure >.<)

at the highest point it was 6.06, using the flats of the calipers so that i didn't dig into the plastic.  Using the tips at one point i saw 6.08 but i think i was just a little unlevel, with that height it is still TIGHT but it will at least seat itself.

I measured the height of the E3D and it was measuring between 5.97 and 5.99 I measured multiple places all the way around it.

clough42, 2015-03-10:
The engineering drawings claim it's 6mm, so that's actually pretty good machining to get it that close.

The overall thickness of the plate, including the top protrusion should be exactly 6mm.  I'm going to print one tonight and see how it comes out for me.  A little die swell is normal, but I wouldn't expect that much.

clough42, 2015-03-10:
Uh oh...mine came out at 5.7mm.  That"s not right.  Looks like I've got some work to do on my printer.

clough42, 2015-03-16:
I think the dimensions on the part are correct (6mm) and the differences we are seeing are differences in printer tuning.

closed 2015-03-16 by clough42