clough42/3d-printing #12: E3DV6 interference with the stock fan screws and X table

Sniffle, 2015-03-10:
the stock fan screws hit the X table preventing a proper fit

pic of the interference point
http://i34.photobucket.com/albums/d130/tourach23/Mobile%20Uploads/2015-03/20150309_212610.jpg

clough42, 2015-03-10:
My intention was for the e3d fan ducts to be installed upside down so the extra fan hangs down in front of the block instead of striking the shelf.  Will they fit the other way around?

If not, I can bevel the front of the shelf.  How much clearance is needed?

Sniffle, 2015-03-10:
I'll check after the kid goes to bed and let you know.  It was probably 2-3
mm

Sniffle, 2015-03-10:
Flipped it over and it fit just fine... was actually a little loose and could turn pretty freely.  

also a pic of the duct with e3d cooling block in place.

i figure this is intentional, just letting you see it in case you want to change anything since this is the first time you've seen it with physical objects
http://i34.photobucket.com/albums/d130/tourach23/Mobile%20Uploads/2015-03/20150309_224107.jpg

clough42, 2015-03-10:
That's just about what I was expecting.  Looks pretty good.  I'm going to close this issue.  I will be very interested to see how much air leaks out of that gap behind the heat sink.  I can tighten it up, but some space is needed on the sides for the fan shroud to fit inside the edges.

closed 2015-03-10 by clough42