clough42/3d-printing #13: Potential clearance issue with part cooling fan duct

Sniffle, 2015-03-15:
I keep having prints fail horribly.   I always start a print and then regularly come back and check on it to make sure it is printing properly.  Every time the print has failed(that i can recall) the print cooling shroud has been flipped up as if it has snagged on a part and dragged.  

I'm going to double check the tightness of the fan duct, to make sure that the duct isn't drooping.  and i'll report back.

As an aside, have you considered adding an M3 nut trap to the part cooling fan to make it easier to tighten the screw when necessary?

Sniffle, 2015-03-16:
ok, so i started a print and noticed something, when priming and wiping the filament, the filament blob that is created when moving from y240 to y0 the blob is catching on the cooling shroud and being bragged along causing it to eventually catch on the print and flip the shroud causing it to then grab onto a part causing the havoc and mayhem.  I'm going to adjust my prime and wipe so that it never runs over the blob, but it was definitely something that was causing me to scratch my head for a bit.

and while not 100% necessary the nut trap would definitely make life a little easier with the cooling shroud but might not look all that great.

clough42, 2015-03-21:
I'm still thinking about it.  I made the fingers for the cooling fan mount narrower so they clear the pulleys better, and this may be why it's slipping.  I may need to make these larger again.

I'm a little reluctant to put in a nut trap because the parts are thin and are printed almost entirely across the print layers.  I think the nut trap would result in a lot of broken parts for folks.  I am considering someday offering these pre-printed, and I think that could generate a lot of warranty issues.

clough42, 2015-03-23:
I'm having no trouble with my fan shroud.  I think I'll close this for now unless more people start having trouble.

closed 2015-03-23 by clough42