clough42/3d-printing #19: Z-screw leveling insufficient

Sniffle, 2015-03-24:
So as i've built tuned and used this I ran across a problem that couldnt be fixed without printing adamfilip's tilt plate due to my Z screws not allowing enough tilt to get the hotends level.   I could get within about .1-.2 mm before the right side of the x carriage would start lifting off the Z nut, and no amount of adjusting would get it to tilt more.  

The only real solutions i can think of are either nut traps that can then force the z axis to crash through the bed, or to provide a tilt plate of some sort.  maybe just referencing adamfillips tilt plate would be sufficient when this is released should it be needed.

clough42, 2015-03-24:
That's interesting.  I wonder why your hot ends are so far off.  Are they different lengths?

The extruder is designed so the hot end should bottom into the extruder block and the groove mount should squeeze it in there, immobilizing it.  If you have any unevenness in the bottoms of the hot end bores, that could cause it as well.

Sniffle, 2015-03-25:
i have no side to side play in them at all and it takes effort to rotate them even a small amount... it might have printed uneven but looking at it it looks dead solid on... I still need to print the updated version and test it... it's just going to be a pain in the butt to do it so i have been putting it off.  

I think on the 12" at least because myself and adamfilip both had the issue, i just don't think we can get as much tilt out of out z screws for some reason.  as to why i don't know, but didn't robh2 also have a similar issue and designed a tilt plate for the older i3 with the steel rods instead of extrudions?

clough42, 2015-03-30:
Not to self...the tilt plate is here:

http://www.thingiverse.com/thing:664362

clough42, 2015-08-02:
I can confirm that the 12" does not tilt as easily as the 8".  I have no idea why.

I just uploaded a new backplate with an additional eccentric on the top for leveling:  http://www.thingiverse.com/thing:945613

This should solve the problem and will be included with the next release of the extruder.

closed 2015-08-02 by clough42