clough42/3d-printing #22: 1.75 flexible filament needs better support

clough42, 2015-05-22:
Need better support structures for 1.75mm filament to prevent the filament from climbing out of the hobbed bolt groove and getting jammed in the mechanism.

clough42, 2015-05-25:
Sketched up a prototype of a block with more support above the bolt.  Provided prototype for testing.

Sniffle, 2015-07-29:
I just installed the latest E3D 1.75 design with the guide hole above the bolt, The center fole for the PTFE was a little small, could be my printer, but the big thing was loading filament was crazy easy with the guide hole above the bolt.  I don't have any flexible to test with, but for standard ABS or PLA it made inerting filament a breeze, no more hunting for the hole under the bolt :-D

clough42, 2015-07-30:
Cool.  I'm glad it's working.  I'm just getting the parts together to build a double 1.75mm E3d setup, and I'm going to use these parts, so I'll be able to adjust the fitment a little better.

clough42, 2015-08-02:
You're right.  The hole for the PTFE was way too tight.  I opened it up another .25mm, and it's now a nice, press fit.

This is fixed and will be in the next release.

closed 2015-08-02 by clough42